edgexfoundry/device-usb-camera #8: Check Caemra Status - 500 Status code -

surajitx-pal, 2022-03-11:
Below the output of this curl command.
$curl -X GET http://localhost:59882/api/v2/device/name/hp-w200-01/CameraStatus |  jq .

**Output :**
{
  "apiVersion": "v2",
  "message": "request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error reading DeviceResourece CameraStatus for hp-w200-01 -\\u003e mandatory query parameter InputIndex not found\",\"statusCode\":500}",
  "statusCode": 500
}

surajitx-pal, 2022-03-13:
logitech c270 usb camera

FelixTing, 2022-03-15:
The /CameraStatus API requires an InputIndex parameter
for example:
$ curl -X GET http://localhost:59882/api/v2/device/name/hp-w200-01/CameraStatus?InputIndex=0

You can find the InputIndex of the camera through this API /GetVideoInputIndex
$ curl -X GET http://localhost:59882/api/v2/device/name/hp-w200-01/GetVideoInputIndex

surajitx-pal, 2022-03-15:
Got it, - What is the input index? - index value will be automatic increment with addition multiple USB cameras?

$ curl -X GET http://localhost:59882/api/v2/device/name/Video_Capture_5-EC5FAE50/GetVideoInputIndex | jq .
{
  "apiVersion": "v2",
  "statusCode": 200,
  "event": {
    "apiVersion": "v2",
    "id": "5ccac4e6-0cd7-4d73-a34e-987240485b83",
    "deviceName": "Video_Capture_5-EC5FAE50",
    "profileName": "HP-Webcam-w200",
    "sourceName": "GetVideoInputIndex",
    "origin": 1647313593795263000,
    "readings": [
      {
        "id": "12b82d6e-042d-4df5-9f26-27b4e972cb9c",
        "origin": 1647313593795263000,
        "deviceName": "Video_Capture_5-EC5FAE50",
        "resourceName": "GetVideoInputIndex",
        "profileName": "HP-Webcam-w200",
        "valueType": "Int32",
        "value": "0"
      }
    ]
  }
}

$ curl -X GET http://localhost:59882/api/v2/device/name/Video_Capture_5-EC5FAE50/CameraStatus?InputIndex=0 | jq .
{
  "apiVersion": "v2",
  "statusCode": 200,
  "event": {
    "apiVersion": "v2",
    "id": "1fbeef26-0072-4130-a1be-5b87aaaf8405",
    "deviceName": "Video_Capture_5-EC5FAE50",
    "profileName": "HP-Webcam-w200",
    "sourceName": "CameraStatus",
    "origin": 1647313662635275500,
    "readings": [
      {
        "id": "e4bd39a9-ae88-4ccc-9921-f8e602e257ff",
        "origin": 1647313662635275500,
        "deviceName": "Video_Capture_5-EC5FAE50",
        "resourceName": "CameraStatus",
        "profileName": "HP-Webcam-w200",
        "valueType": "Uint32",
        "value": "0"
      }
    ]
  }
}

FelixTing, 2022-03-15:
According to the [V4L2 API document](https://www.kernel.org/doc/html/v5.16/userspace-api/media/v4l/vidioc-g-input.html), the index indicates the current number of the video input. Therefore, if the camera has only one video input, the index value will always be 0.

As I see it, the device-usb-camera currently only supports the devices with one camera.
As for the devices with multiple cameras, such as https://www.avermedia.com/us/product-detail/PW313D, we need to support the VIDIOC_S_INPUT V4L2 API in device-usb-camera to select the video input.

surajitx-pal, 2022-03-15:
device-usb-camera -service only supported specific camera brand-model only? Or this will depends in V4l2 intermediate device driver?

FelixTing, 2022-03-15:
It depends on the V4L2 driver. It should support all kinds of UVC(USB Video Class) devices.

lenny-intel, 2023-02-07:
Not an isuue

closed 2023-02-07 by lenny-intel