edgexfoundry/device-usb-camera #9: Start streaming with engative input and output json param - but always return 200 success return code.

surajitx-pal, 2022-03-11:
Negative Test Cases :

$ curl -X PUT http://localhost:59882/api/v2/device/name/hp-w200-01/StartStreaming --header 'Content-Type: application/json' -d '{
>     "StartStreaming": {
>       "InputImageSize": "640x480",
>       "InputPixelFormat" : "YUYV 4:2:2",
>       "OutputVideoQuality": "5",
>       "OutputAspect": "1:1"   **================> Worng input Insead of "4:3" pass "1:1" - but still success**
>     }
> }' 
{"apiVersion":"v2","statusCode":200}

=========================
$ curl -X PUT http://localhost:59882/api/v2/device/name/hp-w200-01/StartStreaming --header 'Content-Type: application/json' -d '{
>     "StartStreaming": {
>       "InputImageSize": "11640x11480", **===================> Wrong Input Still retrun code 200**
>       "InputPixelFormat" : "YUYV 4:2:2",
>       "OutputVideoQuality": "5",
>       "OutputAspect": "0:1"
>     }
> }' 
{"apiVersion":"v2","statusCode":200}

===============================
$ curl -X PUT http://localhost:59882/api/v2/device/name/hp-w200-01/StartStreaming --header 'Content-Type: application/json' -d '{
>     "StartStreaming": {
>       "InputImageSize": "640x480",
>       "InputPixelFormat" : "YUYV 4:2:2",
>       "OutputVideoQuality": "5",
>       "OutputAspect": "4:3",
>       "OutputVideoCodec": "mmpeg444" ==================> Wrong input instead of "mpeg4"
>     }
> }'
{"apiVersion":"v2","statusCode":200}

----------------------------------------
After that check the get parameter command that is keep is default value.
curl -X GET http://localhost:59882/api/v2/device/name/hp-w200-01/GetDataFormat | jq .

"PixelFormat": "YUYV 4:2:2"

cloudxxx8, 2022-03-13:
May I know which camera you are using?  @FelixTing can you see the same result from your camera?

vyshali-chitikeshi, 2022-03-14:
We are using Logitech C270 USB camera for this test.

surajitx-pal, 2022-03-15:
Camera - USB, C270, Logitech,

Steps-1
Auto Discovery - settings , Change the Interval = "30s" at file 'configuration.toml'
UP the all docker services -

Steps-1
Then add Provision Watcher -

curl -X POST \
-d '[
   {
      "provisionwatcher":{
         "apiVersion":"v2",
         "name":"USB-Camera-Provision-Watcher",
         "adminState":"UNLOCKED",
         "identifiers":{
            "Path": "."
         },
         "serviceName": "device-usb-camera",
         "profileName": "USB-Camera-General"
      },
      "apiVersion":"v2"
   }
]' http://localhost:59881/api/v2/provisionwatcher

--------------------------------------------------------
Now start streaming with different input and output (Negative input-output parameters) like "OutputAspect": "1:5" & "InputImageSize": "999x1480" - Later check the streaming status - setting the wrong value and return Status code -200, Below the command output details -
--------------------------------------------------

$ curl -X PUT http://localhost:59882/api/v2/device/name/Video_Capture_5-EC5FAE50/StartStreaming --header 'Content-Type: application/json' -d '{
>     "StartStreaming": {
>       "InputImageSize": "999x1480",
>       "InputPixelFormat" : "YUYV 4:2:2",
>       "OutputVideoQuality": "5",
>       "OutputAspect": "1:5",
>       "OutputVideoCodec": "mpeg4"
>     }
> }' 
{"apiVersion":"v2","statusCode":200}

$ curl -X GET http://localhost:59882/api/v2/device/name/Video_Capture_5-EC5FAE50/StreamingStatus | jq .

{
  "apiVersion": "v2",
  "statusCode": 200,
  "event": {
    "apiVersion": "v2",
    "id": "2e408e97-ded3-4e13-a0b9-24afd89b00c4",
    "deviceName": "Video_Capture_5-EC5FAE50",
    "profileName": "HP-Webcam-w200",
    "sourceName": "StreamingStatus",
    "origin": 1647303856892642600,
    "readings": [
      {
        "id": "308b59a7-27ad-4366-a140-191a57f45847",
        "origin": 1647303856892642600,
        "deviceName": "Video_Capture_5-EC5FAE50",
        "resourceName": "StreamingStatus",
        "profileName": "HP-Webcam-w200",
        "valueType": "Object",
        "objectValue": {
          "InputFps": "30",
          "InputImageSize": "999x1480",
          "IsStreaming": false,
          "OutputAspect": "1:5",
          "OutputFps": "30",
          "OutputFrames": "10",
          "OutputImageSize": "640x480",
          "OutputVideoQuality": "5"
        }
      }
    ]
  }
}

FelixTing, 2022-03-15:
@surajitx-pal Thanks for the detailed information.

Under the hood, the service starts a goroutine to run FFmpeg to stream video frames.
https://github.com/edgexfoundry-holding/device-usb-camera/blob/bdaf71ca9ed1296ecbfb2f1111c7f2781518b11a/internal/driver/driver.go#L513-L539
That's why the service will only return 500 if the device is currently streaming video frames (StreamingStatus.IsStreaming = true). IMO, we can cache [the error message](https://github.com/edgexfoundry-holding/device-usb-camera/blob/bdaf71ca9ed1296ecbfb2f1111c7f2781518b11a/internal/driver/driver.go#L525) in [StreamingStatus](https://github.com/edgexfoundry-holding/device-usb-camera/blob/bdaf71ca9ed1296ecbfb2f1111c7f2781518b11a/internal/driver/device.go#L34-L43) when FFmpeg throws an error so that users are able to know what happened by calling the /StreamingStatus API.

Regarding the wrong input of image size, I noticed that the FFmpeg will automatically pick the closest one from the sizes supported by the device.

Another problem I can see is that the device-usb-camera service currently uses FFmpeg framework to capture video frames and stream them to the RTSP server. The video options mentioned in [README](https://github.com/edgexfoundry-holding/device-usb-camera#advanced-topics) are actually for FFmpeg. If you need to select "YUYV 4:2:2" as input pixel format, set InputPixelFormat to "yuyv422".
This does confuse users. Maybe we should make [the pixel format names provided by go4vl](https://github.com/vladimirvivien/go4vl/blob/main/v4l2/format.go#L33-L42) consistent with FFmpeg?

@cloudxxx8 What's your take on this?

closed 2022-07-26 by lenny-intel