edgexfoundry/device-usb-camera #10: ``` uname -r 5.11.0-1022-aws  sudo find / -name videodev2.h /usr/include/linux/videodev2.h  go version go version go1.17.8 linux/amd64    ```  GCC   ``` gcc -v Using built-in specs. COLLECT_GCC=gcc COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper OFFLOAD_TARGET_NAMES=nvptx-none:hsa OFFLOAD_TARGET_DEFAULT=1 Target: x86_64-linux-gnu Thread model: posix gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04) ```    ```  make CGO_ENABLED=1 GO111MODULE=on go build -ldflags "-X github.com/edgexfoundry/device-usb-camera.Version=0.0.0" -o cmd/device-usb-camera ./cmd # github.com/vladimirvivien/go4vl/v4l2 ../go/pkg/mod/github.com/vladimirvivien/go4vl@v0.0.2-0.20211216162907-40b41ba86c5c/v4l2/capability.go:48:33: could not determine kind of name for C.V4L2_CAP_IO_MC make: *** [Makefile:21: cmd/device-usb-camera] Error 2 ```

wubigo, 2022-03-12:
[https://github.com/vladimirvivien/go4vl/issues/17](https://github.com/vladimirvivien/go4vl/issues/17)

cloudxxx8, 2022-03-13:
@FelixTing please help see whether it is easy to fix and open the PR in https://github.com/vladimirvivien/go4vl if needed

surajitx-pal, 2022-03-13:
Logitech C270 UBS camera

FelixTing, 2022-03-15:
The error might be relevant to Linux kernel version.

There is an open issue https://github.com/vladimirvivien/go4vl/issues/13 related to this.
The author of go4vl says that the minimum kernel version supported is 5.10.x and up. However, I can build binary successfully on 5.10.0 but fail on 5.13.0. Further investigation is needed...

wubigo, 2022-03-16:
reporting error even kernel update to 

```
 uname -r
5.10.5-051005-generic
```

wubigo, 2022-03-16:
sudo dpkg -i linux-libc-dev_5.10.0-14.15_amd64.deb

closed 2022-03-16 by wubigo