edgexfoundry/device-usb-camera #37: gov4l package has a minimum kernel version dependency of 5.10.x

tonyespy, 2022-05-25:
**Version**: this service has not yet been released
**Source**: HEAD (f39f873)

This device service relies on a package named go4vl (https://github.com/vladimirvivien/go4vl) which was recently approved** for usage by the TSC. We ran into problems building this service in a snap package using the `core20` base snap (itself based on Ubuntu 20.04 LTS).

This `gov4l` package is a go wrapper around the Linux kernel V4L2 API. The package compilation requirements include:

 * minimum kernel version required is v5.10.x
 * header files for V4L2 (i.e. /usr/include/linux/videodev2.h)

Since the 20.04 LTS kernel is based on 5.4, we can't build the service on 20.04.

Looking at the Dockerfile for this service, it installs the Alpine `linux-headers` package, although the kernel version of this package isn't explicit, so I can't tell from the Dockerfile what kernel the service is being built against. The referenced kernel header file (`videodev2.h` which defines the V4L2 API) changes across kernel versions. In theory, if we build against a single kernel version (e.g. 5.10), we should be able to use the device service on that version of the kernel or greater, but we should actually validate this.  What distro and version(s) were this device service tested on?

** I question why we approved use of such a new package that hasn't yet even had a single minor version released (this was noted in @farshidtz's review which had requested changes)? The package has only had two releases and the latest tag is v0.0.2!

cloudxxx8, 2022-05-26:
that's the only usable library when we made the research, and the owner help review and merge the PRs whenever we found the issues

cloudxxx8, 2022-06-07:
we just added a line in readme to state the kernel is limited to 5.10.x in #39

iain-anderson, 2022-06-15:
Ideally we should verify at runtime that we have a compatible kernel. This sounds like something the go4vl package should do for us, perhaps raise an issue there?

cloudxxx8, 2022-06-16:
the compatible runtime works fine so far.  we might raise the issue if we identify something is really wrong.

lenny-intel, 2022-07-19:
Addressed by #39

closed 2022-07-19 by lenny-intel