edgexfoundry/device-usb-camera #41: Please update goffmpeg module

tonyespy, 2022-06-10:
This device service has the following dependency on goffmpeg:

 - github.com/xfrr/goffmpeg v0.0.0-20210624103149-5ca2d3062daf

Why are we pointing to a specific git tag of this module???? This module has had 10+ releases, but the last one was released in April of 2020.

Was this module reviewed and approved by the TSC like this?

cloudxxx8, 2022-07-20:
@lenny-intel there is nothing to change here.  the previous release tag is too old.

cloudxxx8, 2022-07-20:
we can open PR to fix issues or fork to our repository if there is any issue we have to fix

lenny-intel, 2022-07-20:
@cloudxxx8 , OK, thanks for looking into this.
@tonyespy , the last code, which is required is not tag for a release. And yes it was approved and is listed on this page.
https://wiki.edgexfoundry.org/display/FA/Vetting+Process+for+3rd+Party+Dependencies

Closing issue as it is what it is.

closed 2022-07-20 by lenny-intel