edgexfoundry/device-usb-camera #48: Allow control of the dockerized RTSP server

farshidtz, 2022-06-29:
# 🚀 Feature Request

### Relevant Package [**REQUIRED**]
Device USB Camera and RTSP Server

### Description [**REQUIRED**]
The Dockerfile uses `aler9/rtsp-simple-server` as a base:
https://github.com/edgexfoundry/device-usb-camera/blob/0d2f133a70ca139fe5a25e6e35809f1e9812dab5/Dockerfile#L36

And runs the process before starting the camera service:
https://github.com/edgexfoundry/device-usb-camera/blob/0d2f133a70ca139fe5a25e6e35809f1e9812dab5/docker-entrypoint.sh#L3-L7

At the same time, the device service allows configuration of the RTSP server:
https://github.com/edgexfoundry/device-usb-camera/blob/0d2f133a70ca139fe5a25e6e35809f1e9812dab5/cmd/res/configuration.toml#L108-L110

I'm not convinced that adding an RTSP server inside the same docker container is a good idea. Note that rtsp-simple-server is already dockerized and easily deployable as standalone: https://hub.docker.com/r/aler9/rtsp-simple-server

There are a few issues:
- The USB Camera service config allows using an external server, but does not allow disabling the internal one. The 9-10MB for the executable gets pulled in regardless.
- It is unclear how someone can configure and secure the internal RTSP server. Does the [upstream documentation](https://github.com/aler9/rtsp-simple-server#table-of-contents) apply to the Device USB service container too? Our README doesn't make that clear.
- Embedding the rtsp-simple-server adds maintenance burden to EdgeX. I don't think this dependency was approved by the TSC. It's been a few weeks and we are already a few versions behind. 

### Describe the solution you'd like
The control of the RTSP server process can be made possible for example by reading an env var in docker-entrypoint.sh to decide if the process should run.

If the RTSP server wasn't embedded, I could easily deploy the project's [container](https://hub.docker.com/r/aler9/rtsp-simple-server) together with the Device Usb Camera service container.

### Describe alternatives you've considered

lenny-intel, 2022-06-29:
@ajcasagrande, thoughts?

cloudxxx8, 2022-06-30:
we don't need to modify the code
it's basically only related to the build script change- Dockerfile and docker-entrypoint.sh
users can build the image together or separately.  that's why we make the RTSP server endpoint configurable in the service
https://github.com/edgexfoundry/device-usb-camera/blob/0d2f133a70ca139fe5a25e6e35809f1e9812dab5/cmd/res/configuration.toml#L108-L110

building them together is simpler for some users, and that's just a reference build script implementation.
users can always modify the build for their own purpose.  I thought we can still use this version for the current release since it's close to the release date, and welcome the contribution to provide another reference build script implementation for the future release.

ajcasagrande, 2022-06-30:
I thought you were saying the `aler9/rtsp-simple-server` is used as the base image, but I see it is just being used to copy the binary and default configuration. 

I think it does make sense to keep it included, and have a flag or something that allows you to disable it.  I'd assume the server can be manually configured by volume mounting over the `rtsp-simple-server.yml` file with a custom one. In fact, all options specified [here](https://github.com/aler9/rtsp-simple-server#configuration) seem valid except for maybe the HTTP API if that port isn't being forwarded.

Side note, the way the Dockerfile is written now should be adjusted to take a build arg of the version and not be hard-coded.

How would a non-bundled one fit in with `edgex-compose`?

farshidtz, 2022-07-15:
> all options specified [here](https://github.com/aler9/rtsp-simple-server#configuration) seem valid except for maybe the HTTP API if that port isn't being forwarded.

Actually, neither of the ports are explicitly exposed in the Dockerfile. RTSP's port is exposed in the reference [docker compose script](https://github.com/edgexfoundry/edgex-compose/blob/ff85f6bff0a3f83f98ac9210ea03f3e4c814f05c/compose-builder/add-device-usb-camera.yml#L24). I think at least the RTSP (8554) port should be exposed in the Dockerfile to make it clearer that another application runs inside the same container.

Also as reported in #51, there are six other servers inside which IMO should be disabled.