edgexfoundry/device-usb-camera #51: Unused rtsp-simple-server listeners in Docker container

farshidtz, 2022-07-15:
The docker image for Device USB Camera pulls in [rtsp-simple-server](https://github.com/aler9/rtsp-simple-server) with its default configurations.

The default configurations results in running the following listeners/servers:
- RTSP (8554)
- RTP (8000)
- RTCP (8001)
- multicastRTP (8002)
- multicastRTCP (8003)
- RTMP (1935)
- HLS (8888)

For more details, refer to: https://github.com/aler9/rtsp-simple-server/blob/main/rtsp-simple-server.yml

The [Dockerfile](https://github.com/edgexfoundry/device-usb-camera/blob/d52eed5e0bdf095fe71ab1c2620e174378e09b3d/Dockerfile) DOES NOT explicitly expose any of the above ports. The servers will still run in containers.

The reference docker compose exposes the RTSP port only:
https://github.com/edgexfoundry/edgex-compose/blob/ff85f6bff0a3f83f98ac9210ea03f3e4c814f05c/compose-builder/add-device-usb-camera.yml#L24

# 🐞 Bug Report

### Affected Services [**REQUIRED**]
Device USB Camera

### Is this a regression?
No

### Description and Minimal Reproduction [**REQUIRED**]
Run a Device USB Camera docker container

## 🔥 Exception or Error
<pre><code>

</code></pre>

## 🌍  Your Environment
**Deployment Environment:**
Docker

**EdgeX Version [**REQUIRED**]:**
v0.0.1-dev.13

**Anything else relevant?**

The snap (see #50) will disable everything other than the RTSP server by setting the following:
```
RTSP_PROTOCOLS: tcp
RTSP_RTSPADDRESS: 127.0.0.1:8554
RTSP_RTMPDISABLE: true
RTSP_HLSDISABLE: true
```

lenny-intel, 2022-07-19:
@cloudxxx8 , can your team address this one?

FelixTing, 2022-07-25:
@farshidtz I have submitted a [PR](https://github.com/edgexfoundry/device-usb-camera/pull/58) for Dockerfile. Could you take care of modifying snapcraft.yml?

farshidtz, 2022-07-25:
Thanks @FelixTing for the fix. We could close this issue which is related to the Docker image only.

We will add a PR to disable the listeners by modifying the config file, instead of setting environment variables.

closed 2022-07-25 by lenny-intel