edgexfoundry/device-usb-camera #64: Device is regiser in edgex metadata after auto-discovery

surajitx-pal, 2022-07-26:
# 🐞 Bug Report
Rest API use to check device list.

curl -X GET http://localhost:59881/api/v2/device/all | jq .

**EdgeX Version [**REQUIRED**]:**
latest- from main branch, edgex-compose repo. V2.

**Anything else relevant?**

after some time ERROR logs in console -

level=INFO ts=2022-07-26T21:04:49.897574986Z app=device-usb-camera source=async.go:104 msg="Adding discovered device UVC_Camera_(046d_0825)-61C0AE50 to Metadata"
level=**ERROR** ts=2022-07-26T21:04:49.901876666Z app=device-usb-camera source=async.go:120 msg="failed to create discovered device UVC_Camera_(046d_0825)-61C0AE50: request failed, status code: 400, err: {\"apiVersion\":\"v2\",\"message\":\"AddDeviceRequest.Device.Name field only allows unreserved characters which are ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789-_~\",\"statusCode\":400}\n"
level=DEBUG ts=2022-07-26T21:04:49.902006461Z app=device-usb-camera source=async.go:127 msg="Filtered device addition finished"

surajitx-pal, 2022-07-26:
Auto-discovery setup, services are up through Edgex Latest main branch(non secure mode). Device is discovered but not showing the device list through edgex command, and not in metadata device profile.

lenny-intel, 2022-07-26:
The device name `UVC_Camera_(046d_0825)-61C0AE50` has the `(` and `)` characters which are not valid for device names.

lenny-intel, 2022-07-26:
Look like the code here needs to include the open/close parenthesis
https://github.com/edgexfoundry/device-usb-camera/blob/fa6315a9dc6abe5875e1ce461ca165c4afc43fa4/internal/driver/device.go#L90-L93

[device.go](https://github.com/edgexfoundry/device-usb-camera/blob/fa6315a9dc6abe5875e1ce461ca165c4afc43fa4/internal/driver/device.go)
func buildDeviceName(cardName, serialNumber string) string {
    cardName = strings.ReplaceAll(strings.ReplaceAll(cardName, " ", "_"), ":", "_")
    serialNumber = strings.ReplaceAll(strings.ReplaceAll(serialNumber, ":", "_"), ".", "_")
    return fmt.Sprintf("%s-%s", cardName, serialNumber)
<https://github.com/[edgexfoundry/device-usb-camera](https://github.com/edgexfoundry/device-usb-camera)|edgexfoundry/device-usb-camera>edgexfoundry/device-usb-camera | Added by [GitHub](https://rsd-intel.slack.com/services/B020N2A516G)

PR #67 (2022-07-27): fix: remove all reserved chars from device name

Closes #64 

**If your build fails**  due to your commit message not passing the build checks, please review the guidelines here: https://github.com/edgexfoundry/edgex-go/blob/main/.github/Contributing.md

## PR Checklist
Please check if your PR fulfills the following requirements:

- [ ] I am not introducing a breaking change (if you are, flag in conventional commit message with `BREAKING CHANGE:` describing the break)
- [ ] I am not introducing a new dependency (add notes below if you are)
- [ ] I have added unit tests for the new feature or b [... cut]
closed 2022-07-27 by ajcasagrande