edgexfoundry/device-usb-camera #89: Intermittently start streaming fails and stream status gives an error

vyshali-chitikeshi, 2022-09-28:
# 🐞 Bug Report
device-usb-camera

### Is this a regression?
No

### Description and Minimal Reproduction [**REQUIRED**]
After executing start streaming api with valid inputs, intermittently stream status gives an error (verified using get stream status api)

## 🔥 Exception or Error
"Error": "Failed Finish FFMPEG ([-nostats -loglevel 0 -y -input_format yuyv422 -s 640x480 -i /dev/video1 -aspect 4:3 -vcodec mpeg4 -qscale 5 -f rtsp -hls_list_size 0 rtsp://localhost:8554/stream/UVC_Camera_046d_0825-F410BE50]) with exit status 1 message 

## 🌍  Your Environment
**Deployment Environment:**

**EdgeX Version [**REQUIRED**]:**

vyshali-chitikeshi, 2022-09-29:
This is happening on Logitech C270 camera

lenny-intel, 2022-10-10:
Make sure passing bad parameters result in error returned to Start streaming. May be related to #23

ajcasagrande, 2022-10-10:
Actually the original message says with valid inputs. More information is required though @vyshali-chitikeshi .

vyshali-chitikeshi, 2022-10-28:
Issue reproducible with images on 10/27/22.
Intermittently start streaming fails even with valid inputs.

Here are input parameters given for starting stream:
{
    "StartStreaming": {
        "InputImageSize": "640x480",
        "InputPixelFormat": "YUYV 4:2:2",
        "OutputVideoQuality": "5",
        "OutputAspect": "4:3",
        "OutputVideoCodec": "mpeg4"
    }
}

Below is the error in streaming status:
{
    "apiVersion": "v2",
    "statusCode": 200,
    "event": {
        "apiVersion": "v2",
        "id": "bb6d9973-0345-4e0c-b2a0-2df5327c7022",
        "deviceName": "UVC_Camera_046d_0825-F410BE50",
        "profileName": "USB-Camera-General",
        "sourceName": "StreamingStatus",
        "origin": 1666928749260786565,
        "readings": [
            {
                "id": "c74430b0-2849-482e-9ca5-2d6f5a78807b",
                "origin": 1666928749260786565,
                "deviceName": "UVC_Camera_046d_0825-F410BE50",
                "resourceName": "StreamingStatus",
                "profileName": "USB-Camera-General",
                "valueType": "Object",
                "value": "",
                "objectValue": {
                    "Error": "Failed Finish FFMPEG ([-nostats -loglevel 0 -y -input_format yuyv422 -s 640x480 -i /dev/video0 -aspect 4:3 -vcodec mpeg4 -qscale 5 -f rtsp -hls_list_size 0 rtsp://localhost:8554/stream/UVC_Camera_046d_0825-F410BE50]) with exit status 1 message  ",
                    "InputFps": "",
                    "InputImageSize": "640x480",
                    "IsStreaming": false,
                    "OutputAspect": "4:3",
                    "OutputFps": "",
                    "OutputFrames": "",
                    "OutputImageSize": "",
                    "OutputVideoQuality": "5"
                }
            }
        ]
    }
}

Below is the one of way to replicate the issue:
1.  Just after setting up the stack, start streaming with valid inputs
2.  Check stream status- 'Is streaming' set to true without any error message
3. Stop stream
4. Check stream status- 'Is streaming' set to false
5.  start stream with invalid inputs - for e.g. with  "OutputVideoCodec": "mpeg45678"
6.  Check stream status- check 'is streaming' set to false with error message
7.  Stop stream
8.  Start streaming with valid inputs

Expected result: Streamign should start correctly and 'is streaming' should set to true
actual result: with valid inputs, intermittently streaming not starting and 'is streaming' set to false

some times, just with starting stream(valid inputs) & stopping stream few times causes the problem.

vyshali-chitikeshi, 2022-11-14:
Issue reproducible intermittently only using my camera so far. We tried with other cameras (different fw/driver version) using same code and its not reproducible. Suspecting its something to do with my camera fw/driver version. I will try with other cameras.

For now closing this ticket as its specific to my Logitech C270 camera with driver version 5.13.19

closed 2022-11-14 by vyshali-chitikeshi