edgexfoundry/device-usb-camera #101: StatusCode 500. device <name> not found

Biiigstone, 2022-10-10:
First of all, I apologize for writing here. I've been struggling with this for weeks, and I couldn't get information from anywhere, so I wrote it.

I followed the guide provided here, but there is an error return from the curl command as follows.

curl -X PUT -d '{
    "StartStreaming": {
      "InputImageSize": "640x480",
      "OutputVideoQuality": "5"
    }
}' http://localhost:59882/api/v2/device/name/Ca/StartStreaming | jq '.'

{
  "apiVersion": "v2",
  "message": "request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error writing DeviceResourece StartStreaming for Ca -\\u003e device Ca not found\",\"statusCode\":500}",
  "statusCode": 500
}

Other commands available via device profile or http://localhost:59882/api/v2/device/all have the same error return. 
(http://localhost:59882/api/v2/device/all?offset=0&limit=20 works strangely)

What should be done to solve this?

Thank you.

lenny-intel, 2022-10-10:
> (http://localhost:59882/api/v2/device/all?offset=0&limit=20 works strangely)

Do you see your expected device in the results in this case?

Biiigstone, 2022-10-11:
1. Same device Ca not found error message? or same error code 500?
=> return the same status code 500 and error message.

2. Do you see your expected device in the results in this case?
=> I'm sorry. I wrote it wrong. That works properly. My intention was to convey 'that works strangely well'.

Since I use a translator, I think I might not be able to communicate well, so I'll explain it with an image.

[image]

[image]
I used Logitec C270 camera and laptop built-in camera.

[image]

I wanted to check it accurately, so I checked it in redis DB.
[image]
[image]
[image]
I think everything is going well.
However, when i command the device, the following error is returned: 'status code 500, device Ca not found'

[image]

The commands that i can obtain in http://localhost:59882/api/v2/device/all?offset=0&limit=20 return the same error and the same message.
[image]

Thank you so much for your interest.

ajcasagrande, 2022-10-12:
@zambranohally @Biiigstone It looks like you have both potentially encountered a bug/limitation of the device service.

Your devices do infact exist in the redis database, which is why `api/v2/devices/all` works for you. This is automatic and would work even when the device service is offline, as this is just telling you the info in the database.

The problem is the device service is sending a misleading error message here:
https://github.com/edgexfoundry/device-usb-camera/blob/e07d484d1f5eb205c2c6f6ecd02e5a0c78a6dd12/internal/driver/driver.go#L617-L627

Which is called by this code during the handling of your REST call:
https://github.com/edgexfoundry/device-usb-camera/blob/e07d484d1f5eb205c2c6f6ecd02e5a0c78a6dd12/internal/driver/driver.go#L160-L163

Essentially the issue boils down to the fact that even though your devices are registered to EdgeX, the device service was unable to (or refused to) connect with that camera and thus it does not add the camera to the known list of "activeDevices". Because of this, all commands will return the same 500 error as you observed.

When a device is added to EdgeX, a callback occurs for the `AddDevice` function. If anything in there returns an error it will end up in that state. A major one is the inability to query the serial number. If you observe the below code snippets, any error generated during this call will cause the system to act in the way you both are experiencing it.

https://github.com/edgexfoundry/device-usb-camera/blob/e07d484d1f5eb205c2c6f6ecd02e5a0c78a6dd12/internal/driver/driver.go#L310-L338

https://github.com/edgexfoundry/device-usb-camera/blob/e07d484d1f5eb205c2c6f6ecd02e5a0c78a6dd12/internal/driver/driver.go#L518-L615

ajcasagrande, 2022-10-12:
@Biiigstone I do not see you mounting `/dev` or `/dev/videoX` into the container via the docker-compose file. Are you missing this?

Biiigstone, 2022-10-13:
I certainly haven't done that process.
Can I get a proper example?
I'm using Ubuntu 20.04 version and I was running it as no-secty.

I'm so happy to think about what will be solved.
Thank you.

ajcasagrande, 2022-10-13:
@Biiigstone I think the documentation here is a bit out of date. Here is what I would recommend:

```bash
# clone a fresh copy of the latest edgex-compose
git clone --recursive https://github.com/edgexfoundry/edgex-compose.git
cd edgex-compose
# checkout latest stable release
git checkout kamakura
# run the edgex stack in no-secty mode with the usb-camera device service
make run ds-usb-camera no-secty
```

You can use `docker ps` to ensure the `device-usb-service` is up.

At this point, you can now add your device as you did before:
[image]

You can use the following command to view the logs for the device service:
```bash
docker logs -f $(docker ps -qf Name="device-usb-camera")
```

Biiigstone, 2022-10-13:
Unfortunately, the method you told me did not work well.

However, I think I got good information from docker logs -f $(docker ps -qf Name="device-usb-camera").

The following is part of my log.

[image]

Thank you.

Biiigstone, 2022-10-13:
[image]

Maybe they solved it!!

I was surprised by the small difference.

Soon, I'll try to recreate it on another computer and if it succeeds, I'll write a note for those who have similar errors as me!

ajcasagrande, 2022-10-13:
Hi @Biiigstone, glad to see it is working for you now! What is the thing that ended up fixing it?

As for the volume mounts, that is handled for you if using the edgex-compose kamakura or newer built-in `ds-usb-service` option to `make run`. Here are the lines for reference:

https://github.com/edgexfoundry/edgex-compose/blob/fa68b85c230c234b7a416225ddaea77d93bf0a4d/compose-builder/add-device-usb-camera.yml#L43-L47

Biiigstone, 2022-10-14:
I don't know the exact reason, but I was able to solve it by running the docker in the root area!! It worked well on other computers, too!

It wasn't the edgeX and usb-device-camera's own problem. I am very sorry about this.

Thank you very much for your interest and time in my problem.

Thank you.

I'll simply write down what I did for someone who's going to have a similar problem with me

1. Get the Device USB Camera Source Code
```
 $ mkdir ~/edgex
 $ cd ~/edgex
 $ git clone --recursive https://github.com/edgexfoundry/device-usb-camera.git
```

2. Download EdgeX Compose Repository (if not already downloaded)
```
 $ cd ~/edgex 
 $ git clone --recursive https://github.com/edgexfoundry/edgex-compose.git
 $ git checkout kamakura
```
3. Deploy EdgeX and USB Device Camera Microservice(sudo!)
 
```
 $ cd ~/edgex/device-usb-camera
 $ sudo make docker
 $ cd ~/edgex/edgex-compose/compose-builder`
 $ sudo make run no-secty ds-usb-camera
``` 

4. Verify Service and Device Profiles
```
 $ sudo docker ps
 $ curl -s http://localhost:59881/api/v2/deviceservice/name/device-usb-camera | jq .
```
 * Status code 200 must be returned.

5. Adding Devices using REST API
* If you use an auto-scanned device with provision watcher, you can skip this step.
* If you want to register your camera yourself, you must delete the auto-registered dv using localhost:4000 or edgex-cli. 
* Because one serial code seems to allow only one dv.

```
 $ v4l2-ctl --list-devices
 $ curl -X POST -H 'Content-Type: application/json'  \          
http://localhost:59881/api/v2/device \
-d '[
         {
            "apiVersion": "v2",
            "device": {
               "name":"Ca",
               "serviceName": "device-usb-camera",
               "profileName": "USB-Camera-General",
               "description": "My test camera",
               "adminState": "UNLOCKED",
               "operatingState": "UP",
               "protocols": {
                  "USB": {
                        "CardName": "NexiGo N930AF FHD Webcam: NexiG",
                        "Path": "/dev/video0",
                                "AutoStreaming": "false"
                  }
               }
            }
         }
]'

``` 
 * Status code 201 must be returned.
 * You should use your name and path.
 * You can use the following command to view the logs for the device service: 
` $ sudo docker logs -f $(sudo docker ps -qf Name="device-usb-camera") `

6. Start Video Streaming

 ```
$ curl -X PUT -d '{
    "StartStreaming": {
      "InputImageSize": "640x480",
      "OutputVideoQuality": "5"
    }
}' http://localhost:59882/api/v2/device/name/<device name>/StartStreaming

``` 
* Status code 200 must be returned.
 
` $ curl -s http://localhost:59882/api/v2/device/name/<device name>/StreamURI | jq '.'`
 * Similarly, under the assumption that status code 200 has been returned, check the value of "value" at the bottom.
 
 For example :
 ```
$ curl -s http://localhost:59882/api/v2/device/name/Ca/StreamURI | jq '.'
{
  "apiVersion": "v2",
  "statusCode": 200,
  "event": {
    "apiVersion": "v2",
    "id": "ee57873d-ef89-46f8-b6fd-5529962cfeb8",
    "deviceName": "Ca",
    "profileName": "USB-Camera-General",
    "sourceName": "StreamURI",
    "origin": 1665710877480039700,
    "readings": [
      {
        "id": "84863745-ea2e-4086-8d65-61510860430e",
        "origin": 1665710877480039700,
        "deviceName": "Ca",
        "resourceName": "StreamURI",
        "profileName": "USB-Camera-General",
        "valueType": "String",
        "value": "rtsp://localhost:8554/stream/Ca"
      }
    ]
  }
}
```

7, Stream the RTSP stream.
` $ ffplay -rtsp_transport tcp <value>`
 
 For example
 `$ ffplay -rtsp_transport tcp rtsp://localhost:8554/stream/Ca `

lenny-intel, 2022-10-17:
@Biiigstone , I am closing this issue as answered. Thank you for the details above.

closed 2022-10-17 by lenny-intel

zambranohally, 2022-11-10:
Hi @ajcasagrande the issue is back again. I don't know what happened.
It is functioning some days ago. Can you help me on this? Did you push and new image for device usb?