edgexfoundry/device-usb-camera #114: Failed to initialize transcoder for device

zambranohally, 2022-11-10:
Hi, I am having issue with usb device. Autodiscovery is working my usb camera is listed in the device, but when I try to call any command for that usb device it always returns 
**{"apiVersion":"v2","message":"request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error reading DeviceResourece CameraStatus for Camera001 -\\u003e device Camera001 not found\",\"statusCode\":500}","statusCode":500}**

And when I open the docker logs, I saw this error below
[image]

Hope somebody here can help me. Thank you in advance
PS. This is working some days ago, I don't know what happened

EdMuNdlm, 2022-11-12:
I'm using a translator, so the grammar may not be smooth. I had a similar problem. I don't know if this is the way to go, but it's worth a try. First, run docker with sudo privileges. Second, the USB-camera-device package is equipped with a provision watcher by default, and automatically detects and registers connected devices when the server starts. If adding a custom device, you must delete the auto-registered device using localhost:4000 (gui) or edgex-cli . There may be a problem caused by a conflict between the device detected by the provision watcher and the custom device. My environment is linux 20.04 LTS, Edge X version is 2.2 thank you

zambranohally, 2022-11-14:
Thank you @EdMuNdlm
But the issue still exists, I follow what you did still failed to initialize the device.
I think this line of code returns error, I don't know what causes it.
[image]

presatish, 2022-11-14:
@zambranohally If the device/camera is auto discovered then it's name should not be `Camera001`, it might be something like for e.g `UVC_Camera_<some-id>` depending on your usb camera.
I think there might be some confusion happening because of the docs. May be you followed this step - https://github.com/edgexfoundry/device-usb-camera/blob/main/docs/RtspStreamingGuide.md#adding-devices-using-rest-api for adding the device? This step is required only if you are adding devices statically/manually. In your case it seems like you want your device to be auto-discovered. So for now run the service as per the docs but skip this step of adding devices statically, if your usb camera is connected the device will be auto-discovered and registered into Edgex. 
You can check the devices registered in Edgex and their names using this command `http://localhost:59881/api/v2/device/all`. Then you can send the command to the respective device. We are working on this service's docs to make it much better and clearer.

zambranohally, 2022-11-15:
@presatish Actually I know the name of my camera device. I query it using curl and also I can check it using the UI via port 4000.
After I run edgex I check the discovered devices using this command "curl http://localhost:59881/api/v2/device/all" and found that my usb camera is listed. so I get the name of my device registered in edgex something like "UHD4K_00...". So I called this command to check the CameraStatus.
**curl http://localhost:59882/api/v2/device/name/<mydevicename>/CameraStatus?IndexInput=0**
it returns 500 status code. It says my camera is not found. I also try to startstreaming still the same error.
So next thing I did is delete the auto discovered device and add it manually and successfully added.
Then I tried again the CameraStatus and StartStreaming endpoint, still it returns not found.
So I clone your device-usb-camera repo and check where the issue reside. I found out that the errror is when initializing the transcoder that is why my device is not listed in the active devices.

zambranohally, 2022-11-15:
The screenshot and error response I shared above are from different camera name. The error response I highlighted above is when I deleted the auto discovered and add it manually and name it Camera001. The screenshot is from the docker logs after edgex starts and discover my usb camera device.

lenny-intel, 2022-11-15:
@zambranohally , what camera model are you using?

zambranohally, 2022-11-15:
@lenny-intel UHD4K
this is the camera I use that is working before
after I follow **Biiigstone** instruction using kamakura version
https://github.com/edgexfoundry/device-usb-camera/issues/101

presatish, 2022-11-15:
@zambranohally Are you still using `kamakura` version for the edgex-compose?

zambranohally, 2022-11-16:
@presatish yes I am currently using kamakura version.

presatish, 2022-11-16:
@zambranohally Edgex recently had a new release, so I believe the latest usb service needs the latest version of edgex-compose. So please do the necessary cleanup of the existing edgex stuff and repeat the steps with the `levski` version of edgex-compose instead of `kamakura`. Hopefully this helps.

zambranohally, 2022-11-17:
Thank @presatish 
I will update my edgex version and try to run usb device again...
Hopefully this works...
Thank for your help

presatish, 2022-11-21:
@zambranohally Is it working now or are you still seeing the issue?

zambranohally, 2022-11-22:
@presatish This is not working on ubuntu 20.04 live but working on ubuntu 20.04 desktop version.
Thank you for your help everyone

lenny-intel, 2022-11-28:
Closed as answered

closed 2022-11-28 by lenny-intel