edgexfoundry/device-usb-camera #209: How to get the streamed video data in app-service-configurable with streaming URI (RTSP URI)?

vediyappan-hcl, 2023-04-03:
I interfaced a webcam with device-usb-camera microservice and able to see the real-time streaming. 

StreamURI: rtsp://localhost:8554/stream/VirtualBox_Webcam_-_HD_User_Fac-<serialnumber>

I need to export streamed video data to a custom application as it is not possible to store binary data in core-data. Before that, I want ensure that the streamed data reaches app-service-configurable. How can I see streamed video data (events/readings) in app-service-configurable?

lenny-intel, 2023-04-03:
@vediyappan-hcl , what are you wanting to do with the `streamed video data` in app-service-configurable?

Note: Normal edge use cases feed the stream into some AI/ML model at the edge. This inference results from the model (typical JSON) are then sent into EdgeX (typically using Device MQTT or Device REST). Then a custom app service can do analysis on the inference data to make decisions.

The following Intel reference implementations demonstrate this approach:
https://intel-iot-devkit.github.io/rtsf-at-checkout-reference-design/
https://intel-iot-devkit.github.io/automated-checkout/index.html

vediyappan-hcl, 2023-04-03:
@lenny-intel Yes, I aware this video inferencing use case. 

I want to store the streamed video data inside EdgeX or outside EdgeX. For that I want to see the video frames in MessageBus or app-service-configurable.

Able to store the video stream outside EdgeX with following command:
`ffmpeg -i rtsp://localhost:8554/stream/VirtualBox_Webcam_-_HD_User_Fac-9464eb7e2d647de4 -vcodec copy -acodec copy -map 0 -f segment -segment_time 300 -segment_format mp4 "ffmpeg_capture-%03d.mp4"`

Inside EdgeX, 1. How can I push the video stream to MessageBus? or 2. How can I see the streamed data in app-service-configurable?

lenny-intel, 2023-04-03:
@vediyappan-hcl , hat are you expecting to do with the video frames in app-service-configurable?

lenny-intel, 2023-04-03:
The camera device services (USB & ONVIF) are not designed to send the video stream through EdgeX.  They are designed to control camera devices from EdgeX. 

I you want to push the stream thru EdgeX (not recommended) you will need to create a service that gets the stream URL from the device service, converts the stream to frames and then pushes those frames into EdgeX via Device MQTT or Device REST.

lenny-intel, 2023-04-10:
@vediyappan-hcl, does this answer your question?

vediyappan-hcl, 2023-04-11:
@lenny-intel Yes, it answered my question. Thank you!

closed 2023-04-11 by vediyappan-hcl