edgexfoundry/device-usb-camera #236: Start streaming API  failing in secure mode

vyshali-chitikeshi, 2023-05-05:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: Device usb service

### Is this a regression?

 No

### Description and Minimal Reproduction [**REQUIRED**]
USB Start streaming API tried to start streaming and then stops immediately in secure mode. Issue observed on Logitech C270 camera. Issue is not reproducible in non-secure mode. 

Repro steps:
1. Deploy the edgex stack and usb service in secure mode with latest code from main branch
2. Verify basic rest-api like getting camera info and camera streaming status works correctly on connected USB camera.
3. Execute start streaming rest-api with valid input &output settings (see attached screenshot)
4. Execute stream status api to verify streaming status.

Expected behavior:
Start streaming starts video streaming and continue streaming until user stops or disconnects camera.
Streaming status api shows 'is streaming' set to true.

Actuak behavior:
Streaming starts for moment and stops immediately. Observed green light coming and going after executing start streaming api.
Streaming status api shows 'is streaming' set to false.

## 🔥 Exception or Error
<pre><code>

</code></pre>
Find attached logs
[start_streaming_failing_usb_secure_mode.txt](https://github.com/edgexfoundry/device-usb-camera/files/11409802/start_streaming_failing_usb_secure_mode.txt)
[image]

## 🌍  Your Environment
**Deployment Environment:**

**EdgeX Version [**REQUIRED**]:**
3.0/latest main

**Anything else relevant?**

ajcasagrande, 2023-05-12:
This is due to https://github.com/edgexfoundry/edgex-go/issues/4565

ajcasagrande, 2023-05-12:
Fixed via https://github.com/edgexfoundry/edgex-compose/pull/378

closed 2023-05-12 by ajcasagrande

vyshali-chitikeshi, 2023-05-12:
Validation  looks good in secure mode.