edgexfoundry/device-usb-camera #263: Panic when device is missing CardName or SerialNumber

presatish, 2023-06-22:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]
device-usb-camera

### Is this a regression?
No

### Description and Minimal Reproduction [**REQUIRED**]

- Run 3.0 version of Edgex core services along with device-usb-camera service (with discovery disabled)
- Add usb camera manually using REST endpoint and omit CardName and/or SerialNumber info while adding. E.g.
      
```
curl -X POST -H 'Content-Type: application/json'  http://localhost:59881/api/v2/device -d '[
    {
    "apiVersion" : "v2",
    "device": {
        "name": "usb-camera",
        "serviceName": "device-usb-camera",
        "profileName": "USB-Camera-General",
        "description": "My test camera",
        "adminState": "UNLOCKED",
        "operatingState": "UP",
        "protocols": {
            "USB": {
                "CardName": "Webcam C270 HD",
                "Path": "/dev/video0",
                "AutoStreaming": "false"
            }
        }
    }
    }
]'
```

- The device is successfully added into Edgex but when we execute some command on the device such as `StartStreaming` the device-usb-camera service panics with the error `panic: interface conversion: interface {} is nil, not string` and eventually exits.

## 🔥 Exception or Error
`panic: interface conversion: interface {} is nil, not string
`
## 🌍  Your Environment
**Deployment Environment:** Docker, Docker + Dev Hybrid

**EdgeX Version [**REQUIRED**]:** 3.0

**Anything else relevant?**
This is fixed in `main`. `CardName` and `SerialNumber` are optional parameters but the service did not have proper checks and when the device was added manually without one or both of these parameters the issue was seen.

lenny-intel, 2023-06-22:
@presatish , the work around for this is to add blank CardName and SerialNumber correct?

lenny-intel, 2023-06-22:
The examples here should have the work around added
https://github.com/edgexfoundry/device-usb-camera/blob/main/cmd/res/devices/general.usb.camera.yaml.example
https://github.com/edgexfoundry/device-usb-camera/blob/main/cmd/res/devices/hp.w200.yaml.example

presatish, 2023-06-22:
> @presatish , the work around for this is to add blank CardName and SerialNumber correct?

There is no need for blank ones, the blank ones will lead to errors. Either the user can add these fields with their respective values or can completely omit them as we have in the examples now.

lenny-intel, 2023-06-23:
@presatish , I am talking about work around for Minnesota, since this is only being fix in Napa.

closed 2023-06-23 by presatish