edgexfoundry/device-usb-camera #320: Not able start Video streaming using device-usb-camera facing Status code :500

divyashree123456, 2024-01-09:
Hi All, 
I m trying to test device-usb-camera device with Edgex Napa version using manual commands in non secure mode.
Was able to run all the services ,device-usb-camera is also running. But facing issue while giving curl command to start streaming.as follows: 
**{"apiVersion":"v3","message":"request failed, status code: 500, err: {\"apiVersion\":\"v3\",\"message\":\"error writing DeviceResource StartStreaming for C922_Pro_Stream_Webcam-FA55FA9F -\\u003e the video streaming process for device C922_Pro_Stream_Webcam-FA55FA9F has stopped -\\u003e failed finish FFMPEG transcoding for device C922_Pro_Stream_Webcam-FA55FA9F (-loglevel level+info -hide_banner -stats_period 60 -y -s 640x480 -i /dev/video2 -qscale 5 -f rtsp -hls_list_size 0 -rtsp_transport tcp rtsp://:@localhost:8554/stream/C922_Pro_Stream_Webcam-FA55FA9F) with exit status 1 message \",\"statusCode\":500}","statusCode":500}**

please do kindly suggest solutions for it.
Thanks in advance

ajcasagrande, 2024-01-09:
@divyashree123456

1. Please make sure you have properly set credentials for the rtsp server. This is a username and password that will be used to protect the outgoing stream for security purposes (even in non-secure mode). Without it, the service will not stream. See [here](https://docs.edgexfoundry.org/3.1/microservices/device/services/device-usb-camera/walkthrough/deployment/#add-credentials-for-the-rtsp-stream) for how to set.

2. Does any other command work such as [CameraStatus](https://docs.edgexfoundry.org/3.1/microservices/device/services/device-usb-camera/supplementary-info/advanced-options/#camerastatus-command) work?

3. Can you please provide the logs for the `device-usb-camera` service?

divyashree123456, 2024-01-10:
Thanks for suggestions ,will add  password and username to RTSP.
Sure will share logs of `device-usb-camera` service

divyashree123456, 2024-01-12:
Hi @ajcasagrande  @lenny-intel ,
Still I m facing "failed finish FFMPEG transcoding for device Integrated_Webcam_HD_Integrate-CN0M05K08LG0019IAVN3A00_Integrated_Webcam_HD "
logs of `device-usb-camera ` service
[image]

Please do suggest to debug

presatish, 2024-01-14:
@divyashree123456 This might be related to outdated version of `ffmpeg` in your system. We had seen this kind of error a while back because the version was old. Make sure the version you are using is 6.0+. Older versions of Ubuntu such as Ubuntu 20.04 might not be able to get newer version via `sudo apt get` in which case you might need to get via adding a new apt repository or downloading binary. Also make sure to clean docker volumes before running all the Edgex services after updating the version.

divyashree123456, 2024-01-16:
Thanks lot @presatish !!
Actually, It was ffmpeg version issue ,earlier it was 3.7, which i updated to 6.1 version.
It started Working once again thanks for all for your suggestions @ajcasagrande @presatish .

closed 2024-01-16 by divyashree123456