edgexfoundry/device-usb-camera #360: rtsp-simple-server executable not found

AkrithiDevadiga, 2024-10-25:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: Dockerfile (unable to pull an image of rtsp-simple-server)

### Description and Minimal Reproduction [**REQUIRED**]

## 🔥 Exception or Error
<pre><code>

level=ERROR ts=2024-10-25T06:17:53.443431456Z app=device-usb-camera source=init.go:86 msg="ProtocolDriver init failed: RtspServerExecutable file cannot be found: stat ./rtsp-simple-server: no such file or directory"
</code></pre>

[image]
[image]

## 🌍  Your Environment

**EdgeX Version [**REQUIRED**]:** Napa

**Anything else relevant?** It is showing in the dockerhub that, aler9/rtsp-simple-server this image is being replaced by https://hub.docker.com/r/bluenviron/mediamtx

[image]

cloudxxx8, 2024-11-13:
it should be fixed in the main branch
https://github.com/edgexfoundry/device-usb-camera/pull/334

AkrithiDevadiga, 2024-11-14:
Yes, but it is being developed for v4 as I can see. What changes should I make to use it for version 3?

cloudxxx8, 2024-11-14:
you can open a PR and make the same change in the napa branch or in your local repository

cloudxxx8, 2025-04-29:
close due to no activity

closed 2025-04-29 by cloudxxx8